Renovus-Tech/solarec-java #62: Inverters - Connect to Solarman

pferrariuy, 2024-09-28:
### API Documentation
- 

### URLs
- Production: 
- Test: 

### Endpoints
- 

### Required information from client / location / generator
- solis.

### Data and data convertion
- Only `DataTypeVo.TYPE_SOLAR_INVERTER_AC_POWER` is available in W (need to divide by 1000)
- Date is in epoch (need to ver converted)
- Data is 1 hour period, need to be splitted to 15 minutes period

Weather information is not availabe and must be retrieve from the `WeatherService`.

### Pendings
- [ ] Determinate data frequency

RomeoIndiaJulietUniform, 2024-09-29:
Hii @pferrariuy, during building the packages for the microservices a part of the tech.renovus.solarec.business.impl.SecurityService is causing problem as the function is not created before. I have commented it as of now , should I create a function for that incomplaince with buisness login ??

[image]

and while converting those DataTypeVo.TYPE_SOLAR_INVERTER_AC_POWER to Watts by dividing by 1000 we will be dealing with Doubles not Integeres so I have also converted the DC power to Double as it will be in complaince while calculations , for which I have changed the interface too. 

[image]

Is there anything I am missing or should I raise a PR after all these changes ?

pferrariuy, 2024-10-01:
Hi @RomeoIndiaJulietUniform nice seeing new faces arround. Thanks for the information.

Currently we are working to implement the service to connect to Solarman API. Do you have any experience in that that could help?

Regarding the issue found at tech.renovus.solarec.business.impl.SecurityServiceImpl, no need to create a new function. Looks like you have an issue in the one of the following places:
- [tech.renovus.solarec.vo.db.relation.DbClientVo](https://github.com/Renovus-Tech/solarec-java/blob/main/solarec.vo/src/main/java/tech/renovus/solarec/vo/db/relation/DbClientVo.java): is where the reference beteween the client and country is stored.
- [tech.renovus.solarec.db.data.dao.base.BaseCountryDao](https://github.com/Renovus-Tech/solarec-java/blob/main/solarec.db/src/main/java/tech/renovus/solarec/db/data/dao/base/BaseCountryDao.java): each DAO file will always work with its corresponding VO, therefore the CountryDao implementations will always return a CountryVo.

Regarding the division by 1000 that will generate Double and not Integer, we are currently working to fix [SonarCloud](https://sonarcloud.io/summary/overall?id=Renovus-Tech_solarec-java) issues. All **security** and **realibility** issues are fixed and working hard to fix **maintainability** (first high, then medium and then low).

Also notice that you have a value of .502, but in reallity must be 502 in the [DataTypeVo](https://github.com/Renovus-Tech/solarec-java/blob/main/solarec.vo/src/main/java/tech/renovus/solarec/vo/db/data/DataTypeVo.java) class. Those are just constant numbers used for IDs. 

Looking closer, looks like you an old version of the proyect and some manual editions have been made, since can't find thoses editions in the blame section of the files.

Fill free to contact is more information is needed.

Regards